OLIMEX/OLINUXINO #19: Debian build instructions incorrect

danielbarry, 2015-01-06:
If you actually follow these commands on the wiki, they don't work.

    # wget https://github.com/OLIMEX/OLINUXINO/blob/master/SOFTWARE/A20/A20-build/spi-sun7i.c

This is because `wget` will retrieve the `html` version of the page, whereas you actually want the `raw` version, like so:

    # wget https://raw.githubusercontent.com/OLIMEX/OLINUXINO/master/SOFTWARE/A20/A20-build/spi-sun7i.c

The other changes required are:

    # wget https://github.com/OLIMEX/OLINUXINO/blob/master/SOFTWARE/A20/A20-build/SPI.patch

to

    # wget https://raw.githubusercontent.com/OLIMEX/OLINUXINO/master/SOFTWARE/A20/A20-build/SPI.patch

and

    # wget https://github.com/OLIMEX/OLINUXINO/blob/master/SOFTWARE/A20/A20-build/a20_olimex_defconfig

to

    # wget https://raw.githubusercontent.com/OLIMEX/OLINUXINO/master/SOFTWARE/A20/A20-build/a20_olimex_defconfig

Other parts of the build may also be affected, I'm yet to get down to those instructions but the same fix will apply.

Thanks for hosting this project by the way, it's really awesome that this exists!

DanKoloff, 2015-01-06:
Thank you for the feedback, danielbarry! These links would be fixed as soon as possible!

bernt, 2016-03-13:
The same error is in https://github.com/OLIMEX/OLINUXINO/blob/master/SOFTWARE/A20/A20-build-3.4.103/A20_Olimex_kernel_3.4.103%2B_Jessie_with_camera_support_building_description.txt

DanKoloff, 2016-03-21:
It doesn't seem that there is any error, bernt. All the links to GitHub in the instructions of the Jessie build contain "/raw/master/".

I believe the links are alright.

closed 2016-03-21 by DanKoloff