OLIMEX/OLINUXINO #23: olimex micro doesnt start

dmsherazi, 2015-05-18:
when the board is powered on it displays this message on the debug uart continuously!

    U-Boot SPL 2014.04-10706-g36080eb (Aug 19 2014 - 16:42:16)
    Board: A20-OLinuXino_MICRO
    DRAM:Timeout initialising DRAM

Please help how can I debug this

TsvetanUsunov, 2015-05-18:
is this our board or yours with different memory chips?

dmsherazi, 2015-05-19:
its a olimex micro board

DanKoloff, 2015-05-19:
Hello, this doesn't seem like a software issue or problem with the default image.

Try to boot with external power supply. USB powering might be insufficient causing certain hardware to remain un-powered.

Make sure you are using the suitable image for your board. Different boards have different images.

If the problem persists contact the customer support at support@olimex.com - it is a good idea to provide more details - which board exactly do you have; did it behave like that since day one; did you try android or another image; how do you power the board. Anything that might us help replicate the problem is welcome.

closed 2015-05-19 by DanKoloff

dmsherazi, 2015-06-02:
I had a problem with power... fixed it!