OLIMEX/OLINUXINO #29: can't cross build python packages

ricardocrudo, 2015-08-11:
I can't build python packages using buildroot. In the building process the following message is received:

```
>>> python-olimex 0.2.1 Patching
>>> python-olimex 0.2.1 Configuring
>>> python-olimex 0.2.1 Building
(cd /home/crudo/modduo-build/ROOT/build/python-olimex-0.2.1//; PATH="/home/crudo/modduo-build/ROOT/host/bin:/home/crudo/modduo-build/ROOT/host/sbin:/home/crudo/modduo-build/ROOT/host/usr/bin:/home/crudo/modduo-build/ROOT/host/usr/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games" PYTHONPATH="/home/crudo/modduo-build/ROOT/target/usr/lib/python3.4/sysconfigdata/:/home/crudo/modduo-build/ROOT/target/usr/lib/python3.4/site-packages/" _python_sysroot=/home/crudo/modduo-build/ROOT/host/usr/arm-buildroot-linux-gnueabihf/sysroot _python_prefix=/usr _python_exec_prefix=/usr  /home/crudo/modduo-build/ROOT/host/usr/bin/python setup.py build  )
running build
running build_py
creating build
creating build/lib.linux-x86_64-3.4
creating build/lib.linux-x86_64-3.4/pyA20
copying pyA20/__init__.py -> build/lib.linux-x86_64-3.4/pyA20
creating build/lib.linux-x86_64-3.4/pyA20/gpio
copying pyA20/gpio/__init__.py -> build/lib.linux-x86_64-3.4/pyA20/gpio
running build_ext
No processor detected
Warning! Detected and target processor mismatch. 
Do you want to continue [Y/n]?
```

nardist01, 2016-03-05:
same problem

SelfDestroyer, 2016-03-07:
Since you're cross compiling the setup script is detecting that the host cpu is not same as the target(AW A20). Because of that you get input form. To bypass this, you should comment some code:

    def print_warning():
    """
    Print confirmation dialog
    :return:
    """
    print (print_color("Warning! ") + "Detected and target processor mismatch. ")

    # var = input("Do you want to continue [Y/n]? ")
    # if var == 'Y' or var == 'y':
    #     return
    # else:
    #     print ("Abort.")
    #    sys.exit(1)

This is **setup.py** file. After that it should work

closed 2016-05-25 by DanKoloff