OLIMEX/OLINUXINO #31: Terrible performance with gigabit ethernet on A20-OLinuXIno-LIME2

dariox86, 2015-12-03:
When gigabit ethernet is enabled by default on Debian images from Olimex, Debian images from the official Debian website and Arch Linux, ethernet performance is terrible. Maximum download speed is 15 KB/s and sometimes drops as low as 5 KB/s. Forcing 100 Mb/s ethernet via softwore (mii-tool eth0 -F 100baseTx-FD) or connecting the board to a 100 Mb/s switch is a workaround, though I have bought the board specifically for the gigabit capability. The problem is mentioned here: http://forum.armbian.com/index.php/topic/31-problems-with-gigabit/ and here: https://www.olimex.com/forum/index.php?topic=4615.0 . The problem is very easy to reproduce: just hook up the board to a 1000 Gb/s network switch and try to download something from the Internet.

MontiMcMannus, 2016-01-28:
It depends on the board, some are ok, some not. You may try to send it back, another board may be useable...

dariox86, 2016-01-28:
Thank you for the answer. Do you work at Olimex? Do I have to open a RMA to send it back? The board was bought several months ago. Is it still under warranty?

dan-cristian, 2016-03-16:
I have the same issue. I run the test between an Intel machine running linux, with gigabit ethernet, and Lime2.
My switch is a 24-port gigabit. Clearly no problem with my network equipment or cables as I have several PCs in my network with gigabit links working OK.

I run the Vanilla Jessie image from http://www.armbian.com/olimex-lime-2/.

DOWNLINK seems fine, getting max 701 Mbits/sec. See below.
UPLINK is terrible, avg. is 3 Mbits/s.

-------------------DOWNLINK------------------------------------
root@lime2:~# iperf3 -c 192.168.0.11
Connecting to host 192.168.0.11, port 5201
[  4] local 192.168.0.27 port 58414 connected to 192.168.0.11 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  59.1 MBytes   495 Mbits/sec    0    204 KBytes
[  4]   1.00-2.02   sec  69.5 MBytes   575 Mbits/sec    0    335 KBytes
[  4]   2.02-3.00   sec  65.4 MBytes   557 Mbits/sec    0    506 KBytes
[  4]   3.00-4.02   sec  65.5 MBytes   541 Mbits/sec    0    765 KBytes
[  4]   4.02-5.00   sec  77.5 MBytes   660 Mbits/sec    0    765 KBytes
[  4]   5.00-6.00   sec  79.6 MBytes   666 Mbits/sec    0   1.02 MBytes
[  4]   6.00-7.00   sec  81.2 MBytes   683 Mbits/sec    0   1.33 MBytes
[  4]   7.00-8.01   sec  78.8 MBytes   656 Mbits/sec    0   1.33 MBytes
[  4]   8.01-9.00   sec  77.5 MBytes   653 Mbits/sec    0   1.50 MBytes
[  4]   9.00-10.01  sec  83.8 MBytes   701 Mbits/sec    0   1.50 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.01  sec   738 MBytes   619 Mbits/sec    0             sender
[  4]   0.00-10.01  sec   738 MBytes   619 Mbits/sec                  receiver

--------------------UPLINK------------------------------------------------

root@lime2:~# iperf3 -c 192.168.0.11 -R
Connecting to host 192.168.0.11, port 5201
Reverse mode, remote host 192.168.0.11 is sending
[  4] local 192.168.0.27 port 58418 connected to 192.168.0.11 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   650 KBytes  5.33 Mbits/sec
[  4]   1.00-2.00   sec   431 KBytes  3.53 Mbits/sec
[  4]   2.00-3.00   sec   291 KBytes  2.39 Mbits/sec
[  4]   3.00-4.00   sec   505 KBytes  4.14 Mbits/sec
[  4]   4.00-5.00   sec   535 KBytes  4.38 Mbits/sec
[  4]   5.00-6.00   sec   240 KBytes  1.97 Mbits/sec
[  4]   6.00-7.00   sec   181 KBytes  1.48 Mbits/sec
[  4]   7.00-8.00   sec  77.8 KBytes   637 Kbits/sec
[  4]   8.00-9.00   sec   573 KBytes  4.69 Mbits/sec
[  4]   9.00-10.00  sec   216 KBytes  1.77 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  3.63 MBytes  3.05 Mbits/sec  373             sender
[  4]   0.00-10.00  sec  3.61 MBytes  3.03 Mbits/sec                  receiver

dariox86, 2016-03-17:
I see. Mine is even worse.

mhaas, 2016-03-17:
There is some work being done to get this fixed via u-boot: https://www.mail-archive.com/u-boot@lists.denx.de/msg206626.html

Please disregard my initial patch there.

dariox86, 2016-03-17:
Wow, this is good news! Thank you for pointing this out.

mhaas, 2016-03-20:
We have something figured out. If you'd like to test a preliminary patch, go check the mailing list.

dariox86, 2016-03-20:
I'd like to. Exactly what is the patch and what is the patch to disregard?

mhaas, 2016-03-20:
This one: https://www.mail-archive.com/u-boot@lists.denx.de/msg207006.html

It's a hack, but it works for the lime2 for me.

I also need the following patch or the axp209/mv264xxx linux modules will lock up the board. YMMV.

````
commit d95334931b464c090e34325a226cd3d1f1cb0f3c
Author: Michael Haas <haas@computerlinguist.org>
Date:   Sun Mar 20 16:59:54 2016 +0100

    sunxi: power: Enable LDO3 and LDO4 again
    
    This reverts commit 02cc27c74f9b884b538bcd1b93342a4c05b5d608.
    
    This revert is required to avoid immediate hangups on
    Debian Jessie with 4.3.0-0.bpo.1-armmp-lpae on loading
    i2c-mv64xxx.
    
    Signed-off-by: Michael Haas <haas@computerlinguist.org>

diff --git a/drivers/power/Kconfig b/drivers/power/Kconfig
index adc6455..110e111 100644
--- a/drivers/power/Kconfig
+++ b/drivers/power/Kconfig
@@ -160,7 +160,7 @@ config AXP_ALDO2_VOLT
 config AXP_ALDO3_VOLT
        int "axp pmic (a)ldo3 voltage"
        depends on AXP209_POWER || AXP221_POWER || AXP818_POWER
-       default 0 if AXP209_POWER
+       default 2800 if AXP209_POWER
        default 3000 if MACH_SUN6I || MACH_SUN8I
        ---help---
        Set the voltage (mV) to program the axp pmic aldo3 at, set to 0 to
@@ -173,7 +173,7 @@ config AXP_ALDO3_VOLT
 config AXP_ALDO4_VOLT
        int "axp pmic (a)ldo4 voltage"
        depends on AXP209_POWER
-       default 0 if AXP209_POWER
+       default 2800 if AXP209_POWER
        ---help---
        Set the voltage (mV) to program the axp pmic aldo4 at, set to 0 to
        disable aldo4.
````

dariox86, 2016-03-27:
Tested your patch on my A20-OLinuXIno-LIME2. You are the man. It works.

dan-cristian, 2016-03-27:
great news. What is the speed (in both directions) you now get after applying the patch?

dariox86, 2016-03-27:
I get 600 Mb/s average in both directions between my Intel x86 macchine and my Lime2 connected through a 1000 Mb/s ethernet router (Vodafone Station Devolution, an ADSL home gateway).

dariox86, 2016-03-27:
Vodafone Station Revolution, sorry.

dan-cristian, 2016-03-27:
Thanks. Do you have by chance a quick guide / steps with how I can recompile my kernel and apply this patch myself?

dariox86, 2016-03-27:
You must recompile the bootloader, not the kernel. What operating system do you use on your client machine?

dan-cristian, 2016-03-27:
I use Vanilla Jessie image from http://www.armbian.com/olimex-lime-2/, http://mirror.igorpecovnik.com/Armbian_4.5_Lime2_Debian_jessie_4.2.3.zip

Thanks
Dan

dariox86, 2016-03-27:
Yes, I know, I read your previous post. I mean, what operating system do you use on your desktop system?

dariox86, 2016-03-27:
If it easier for you, you can cross-compile the bootloader on your desktop computer.

dan-cristian, 2016-03-27:
I have several OS-es at home I can use:
- Windows 10 Enterprise 64 bit
- Windows 8.1 64bit
- Debian Intel (Linux nas 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt9-3~deb8u1
(2015-04-24) x86_64 GNU/Linux)

dariox86, 2016-03-27:
I will prepare a quick guide for you to build it on Debian.

dan-cristian, 2016-03-27:
many thanks!

igorpecovnik, 2016-03-27:
I already added a patch to Armbian build script. Haven't test but it should work.

dan-cristian, 2016-03-27:
Thanks. I can test that as well if you point me to the build instructions.

igorpecovnik, 2016-03-27:
Start here:
https://github.com/igorpecovnik/lib

You only need one switch, KERNEL_ONLY="yes" ... i guess you only need u-boot and kernel.

For advanced features, check here:
https://github.com/igorpecovnik/lib/blob/master/documentation/geek-faq.md

dariox86, 2016-03-27:
@dan-cristian, it may be easier for you to follow @igorpecovnik route.

dan-cristian, 2016-03-27:
OK, I am trying that, I hoped your suggestion will be quicker. Thanks.

dariox86, 2016-03-27:
OK. I will write a quick 'n dirty guide for you. Just give me some minute.

dariox86, 2016-03-27:
Be warned that this is just a draft to get you up and running in short time. Also, bear in mind that you may render your microSD unbootable. There is definitely a leaner way to accomplish the task. For example, many packages may not be needed. You just wanted a quick 'n dirty solution to try it out. Here it is.

- Install the build environment:
`#	apt-get install git device-tree-compiler debconf-utils pv bc lzop zip binfmt-support bison build-essential ccache debootstrap flex gawk gcc-arm-linux-gnueabihf lvm2 qemu-user-static u-boot-tools uuid-dev zlib1g-dev unzip libusb-1.0-0-dev parted pkg-config expect gcc-arm-linux-gnueabi libncurses5-dev git vim screen`

- Download the source code for U-Boot:
`$	git clone https://github.com/RobertCNelson/u-boot`

- Replace the files in u-boot/drivers/net/phy and u-boot/drivers/power with the attached ones:
`[source.zip](https://github.com/OLIMEX/OLINUXINO/files/191066/source.zip)`

- Enter the directory:
`$	cd u-boot/`

- Build the source:
```
$	make -s CROSS_COMPILE=arm-linux-gnueabihf- clean
$	make A20-OLinuXino-Lime2_defconfig CROSS_COMPILE=arm-linux-gnueabihf-
$	make CROSS_COMPILE=arm-linux-gnueabihf-
```

- Insert your microSD card and overwrite the patched U-Boot (where sdX is your microSD card):
`#	dd if=u-boot-sunxi-with-spl.bin of=/dev/sdX bs=1024 seek=8`

There is a significant risk I missed something. You may render your microSD unbootable. Let me know if it works for you.

dan-cristian, 2016-03-27:
thanks, I'll try and let you know. my lime2 has a clean install so no
worries about making SD unbootable.

mhaas, 2016-03-27:
Just out of curiosity: why are you not using mainline u-boot, E.g 2016.01 or 2016.3 (which requires an additional patch)?

dariox86, 2016-03-27:
I just found this repository somewhere on the Internet. As I said, it's just a dirty solution.

dan-cristian, 2016-03-27:
I have tested first the solution provided by @igorpecovnik, thanks Igor!

I've built a complete image on jessie, using "next" kernel.
root@lime2:~# uname -a: Linux lime2 4.4.6-sunxi #2 SMP Sun Mar 27 22:01:23 BST 2016 armv7l GNU/Linux

Lime Upload is 554 Mbit/s
Lime download is 107 Mbit/s ---- BETTER, but not as it should be. Any idea on how to increase this?

Ping packets (9000 bytes size) are not lost, looking good here.

Details below:

root@lime2:~# iperf3 -c 192.168.0.11
Connecting to host 192.168.0.11, port 5201
[  4] local 192.168.0.23 port 36338 connected to 192.168.0.11 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.02   sec  67.1 MBytes   554 Mbits/sec    0    464 KBytes
[  4]   1.02-2.01   sec  65.0 MBytes   547 Mbits/sec    0    604 KBytes
[  4]   2.01-3.01   sec  65.0 MBytes   548 Mbits/sec    0    604 KBytes
[  4]   3.01-4.01   sec  65.0 MBytes   541 Mbits/sec    0    666 KBytes
[  4]   4.01-5.01   sec  65.0 MBytes   547 Mbits/sec    0    700 KBytes
[  4]   5.01-6.01   sec  65.0 MBytes   546 Mbits/sec    0    700 KBytes
[  4]   6.01-7.00   sec  65.0 MBytes   548 Mbits/sec    0    700 KBytes
[  4]   7.00-8.00   sec  66.2 MBytes   557 Mbits/sec    0    741 KBytes
[  4]   8.00-9.00   sec  70.0 MBytes   587 Mbits/sec    0    778 KBytes
^C[  4]   9.00-9.07   sec  5.25 MBytes   657 Mbits/sec    0    778 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-9.07   sec   599 MBytes   554 Mbits/sec    0             sender
[  4]   0.00-9.07   sec  0.00 Bytes  0.00 bits/sec                  receiver

root@lime2:~# iperf3 -c 192.168.0.11 -R
Connecting to host 192.168.0.11, port 5201
Reverse mode, remote host 192.168.0.11 is sending
[  4] local 192.168.0.23 port 36342 connected to 192.168.0.11 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  8.53 MBytes  71.6 Mbits/sec
[  4]   1.00-2.00   sec  13.2 MBytes   111 Mbits/sec
[  4]   2.00-3.00   sec  10.1 MBytes  84.3 Mbits/sec
[  4]   3.00-4.00   sec  18.7 MBytes   157 Mbits/sec
[  4]   4.00-5.00   sec  10.4 MBytes  87.3 Mbits/sec
[  4]   5.00-6.00   sec  11.9 MBytes  99.9 Mbits/sec
[  4]   6.00-7.00   sec  18.4 MBytes   154 Mbits/sec
[  4]   7.00-8.00   sec  13.0 MBytes   109 Mbits/sec
[  4]   8.00-9.00   sec  10.0 MBytes  84.1 Mbits/sec
[  4]   9.00-10.00  sec  13.5 MBytes   114 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec   128 MBytes   107 Mbits/sec  2509             sender
[  4]   0.00-10.00  sec   128 MBytes   107 Mbits/sec                  receiver

dariox86, 2016-03-27:
I get around 700 Mb/s downlink and 500 MB/s uplink.

igorpecovnik, 2016-03-28:
@dariox86 Strange, I used the same approach as found in your patch. Which version of u-boot do you use? Latest development?

Next thing is to play around with CONFIG_GMAC_TX_DELAY parameter.

mhaas, 2016-03-28:
@igorpecovnik I may be missing some context here - are there any outstanding issues?

dan-cristian, 2016-03-28:
Yes, I run the compile process using the latest version from github,
exactly as per your install steps.

I have first tried to install only u-boot.deb and image.deb packages, but
my board did not boot the kernel after that, so I had to compile the
complete image.

If I change that parameter, how do I quickly compile and apply the patch,
without creating the whole image again?

Thanks,
Dan

igorpecovnik, 2016-03-28:
@mhaas I am doing this blindly. Don't know. I only added a patch directly to lime2 config:
https://github.com/igorpecovnik/lib/blob/master/patch/u-boot/u-boot-next/lime2-enable-ldo4.patch
which I guess is needed in latest mainline uboot.

@dan-cristian 

Quicker is manual u-boot compilation and installing a boot loader directly with DD. You can also compile native, on Lime, it does not take so long.

dariox86, 2016-03-28:
@igorpecovnik, "my" patch is @mhaas patch. I just applied it to the repository you see in my post. Are you still having severe speed issues with it?

mhaas, 2016-03-28:
@igorpecovnik the LDO3/LDO4 patch fixes a different issue where I/O ports E and G are disabled when they shouldn't be. Not an issue in itself, but re-enabling them causes hangs on my system.. different issue, I am not sure how that got mixed up.

Anyways, you will need ALL THREE patches of the following series:
https://patchwork.ozlabs.org/patch/602067/
AND
http://git.denx.de/?p=u-boot.git;a=commit;h=1eae8f66ff749409eb96e2f3f3387c56232d0b8a
AND
http://git.denx.de/?p=u-boot.git;a=commit;h=fc8991c61c393ce6a9d3dfc97cb56dbbd9e8cbba

On top of u-boot 2016.03. If you are using 2016.01, you don't need the last patch.

igorpecovnik, 2016-03-28:
OK, got it, mess is bigger than I thought. I already added one a patch from Hans, a week ago or so:
http://git.denx.de/?p=u-boot.git;a=commit;h=fc8991c61c393ce6a9d3dfc97cb56dbbd9e8cbba

Thanks for update.

dan-cristian, 2016-04-02:
I have also received a patch from Olimex that improves the speed and removes ping losses.
See test results below.

Patch and instructions can be downloaded from here: 
https://drive.google.com/file/d/0B7WHuNCASY8cUF85UEZWTGltRG8/view?usp=sharing
The above u-boot works for both A20-OLinuXino-LIME2 and A20-SOM-EVB.

root@lime2:~# iperf3 -c 192.168.0.9 -R
Connecting to host 192.168.0.9, port 5201
Reverse mode, remote host 192.168.0.9 is sending
[  4] local 192.168.0.114 port 52134 connected to 192.168.0.9 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  66.8 MBytes   559 Mbits/sec
[  4]   1.00-2.00   sec  76.9 MBytes   645 Mbits/sec
[  4]   2.00-3.00   sec  77.1 MBytes   647 Mbits/sec
[  4]   3.00-4.00   sec  77.2 MBytes   648 Mbits/sec
[  4]   4.00-5.00   sec  77.0 MBytes   647 Mbits/sec
[  4]   5.00-6.00   sec  76.9 MBytes   643 Mbits/sec
[  4]   6.00-7.00   sec  75.9 MBytes   637 Mbits/sec
[  4]   7.00-8.00   sec  75.8 MBytes   637 Mbits/sec
[  4]   8.00-9.00   sec  76.8 MBytes   642 Mbits/sec
[  4]   9.00-10.00  sec  76.0 MBytes   638 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec   761 MBytes   638 Mbits/sec    6             sender
[  4]   0.00-10.00  sec   756 MBytes   634 Mbits/sec                  receiver

iperf Done.
root@lime2:~# iperf3 -c 192.168.0.9
Connecting to host 192.168.0.9, port 5201
[  4] local 192.168.0.114 port 52138 connected to 192.168.0.9 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.01   sec  58.7 MBytes   487 Mbits/sec    0    293 KBytes
[  4]   1.01-2.01   sec  64.0 MBytes   540 Mbits/sec    0    390 KBytes
[  4]   2.01-3.02   sec  65.0 MBytes   540 Mbits/sec    0    458 KBytes
[  4]   3.02-4.01   sec  63.8 MBytes   540 Mbits/sec    0    481 KBytes
[  4]   4.01-5.01   sec  65.0 MBytes   543 Mbits/sec    0    587 KBytes
[  4]   5.01-6.02   sec  65.0 MBytes   542 Mbits/sec    0    735 KBytes
[  4]   6.02-7.00   sec  63.8 MBytes   543 Mbits/sec    0    772 KBytes
[  4]   7.00-8.01   sec  65.0 MBytes   541 Mbits/sec    0    772 KBytes
[  4]   8.01-9.02   sec  65.0 MBytes   541 Mbits/sec    0    772 KBytes
[  4]   9.02-10.01  sec  63.8 MBytes   539 Mbits/sec    0    772 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.01  sec   639 MBytes   536 Mbits/sec    0             sender
[  4]   0.00-10.01  sec   639 MBytes   536 Mbits/sec                  receiver

dariox86, 2016-04-02:
How did you manage to get them to sent you this patch? The source code for it may be useful.

mhaas, 2016-04-03:
It's likely this set of files: https://www.olimex.com/forum/index.php?topic=4615.msg21467#msg21467

dan-cristian, 2016-04-04:
Yes, is that set of files indicated by Michael. I contacted Olimex support via email and they were very helpful and delivered the patch.

BTW, the Lime2 CPU stays at 99% while iperf-ing in average with 600 Mbits, so I wonder if the board can actually deliver faster network speeds (~900 mbit)? Is there any other optimisation that can be done?

closed 2016-05-24 by DanKoloff