OLIMEX/OLINUXINO #36: free space new release openwrt

ebdavide, 2016-04-08:
I have just updated the firmware of my rt5350f to the last release of chaos calmer 
https://downloads.openwrt.org/chaos_calmer/15.05.1/ramips/rt305x/openwrt-15.05.1-ramips-rt305x-olinuxino-rt5350f-squashfs-sysupgrade.bin
After having finished the installation via tftp I noticed that the free space is now just 232 KB, while it was 2 MB with the previous version. Is there any module or packet that has been added and that I can remove. Do you think I did something wrong?

ebdavide, 2016-04-08:
On the current version of chaos calmer 15.05.1 when I type df -h I obtain:
> root@OpenWrt:~# df -h
> Filesystem                Size      Used Available Use% Mounted on
> rootfs                  448.0K    216.0K    232.0K  48% /
> /dev/root                 2.3M      2.3M         0 100% /rom
> tmpfs                    14.2M    624.0K     13.6M   4% /tmp
> tmpfs                    14.2M     48.0K     14.2M   0% /tmp/root
> tmpfs                   512.0K         0    512.0K   0% /dev
> /dev/mtdblock6          448.0K    216.0K    232.0K  48% /overlay
> overlayfs:/overlay      448.0K    216.0K    232.0K  48% /

On the previous version instead:
> root@OpenWrt:/# df -h
> Filesystem                Size      Used Available Use% Mounted on
> rootfs                    4.2M      2.1M      2.1M  50% /
> /dev/root                 2.5M      2.5M         0 100% /rom
> tmpfs                    14.3M    432.0K     13.8M   3% /tmp
> /dev/mtdblock6            4.2M      2.1M      2.1M  50% /overlay
> overlayfs:/overlay        4.2M      2.1M      2.1M  50% /
> tmpfs                   512.0K         0    512.0K   0% /dev

SelfDestroyer, 2016-04-11:
The previous version is the one from Olimex?

ebdavide, 2016-04-11:
yes, the previous comes from the olimex repository, the new one is downloaded from the openwrt website

zgyarmati, 2018-05-27:
Since this the partition size was fixed in upstream OpenWrt as well, see commit 
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=88765756013548cc86cb95c585e22079622b588c,

so i think this can be closed

DanKoloff, 2018-05-28:
Nice @zgyarmati ! Closing the issue.

closed 2018-05-28 by DanKoloff