OLIMEX/OLINUXINO #39: Default SPI speed is much too slow

martinayotte, 2016-08-11:
Default SPI speed is much too slow !

I've an SPI LCD that was working fine in kernel 4.4.5, and suddenly was way too slow when I've upgraded to 4.5.x or 4.6.x. After some debugging, I've figured out that kernel was now using "transfer-speed" instead of "spi-device-max-speed", the first been set during ioctrl, but ignored/not-used in older kernels.

Here is the commit merge in Mainline 9 months ago : 

https://github.com/megous/linux/commit/47284e3e0f3c427c93f8583549b6c938e8a18015

I've finally figured out that guilty code was in pyA20 providing a wrong default ioctrl value for the "transfer-speed" of 100kHz, while we all know that SPI default should be 10MHz : 

`diff pyA20-0.2.1/pyA20/spi/spi.c~ pyA20-0.2.1/pyA20/spi/spi.c
238c238
<     config.speed = 100000;
---
>     config.speed = 10000000;
`

FahriAyberk, 2016-09-20:
Dear martinayotte

I'm quite new to this board and Kernel (Olimex Micro A20) and have been trying to use this Python package to communicate through SPI for a while now.

Whenever I issue the spi.read() function Python aborts with the message below 

`>>> spi.read(2)
*** stack smashing detected ***: python terminated
Aborted
`
My Image Is:
 ARMBIAN Debian GNU/Linux 8 (jessie) 3.4.112-sun7i

Even though I've tried many images and read many forums I couldn't figure out why? Since you have this issue with speed rather recently, I have hoped that you might already passed my level of problem.

martinayotte, 2016-10-17:
@FahriAyberk , sorry, I've no idea ...

DanKoloff, 2017-04-19:
Thanks for sharing your solution about the SPI speed. This issue is now closed.

closed 2017-04-19 by DanKoloff

martinayotte, 2017-04-19:
@DanKoloff , but there is no commit about this fix in pyA20 yet ...

DanKoloff, 2017-04-19:
Yes. But there is no pyA20 in the official images at all. Neither does the official images use Kernel 4.4.5. So I see how this information can be useful if somebody encounters something similar and that is fine. I fail to see how this is an issue for this specific repository.

martinayotte, 2017-04-19:
The fix should be applied there : 
https://github.com/OLIMEX/OLINUXINO/blob/master/SOFTWARE/Python/pyA20/pyA20/spi/spi.c#L223

DanKoloff, 2017-04-20:
Oh, this code is outdated, it has more bugs even. It was uploaded at GitHib first, before starting committing in PyPi. The updated one is here: https://pypi.python.org/pypi/pyA20/0.2.1

Thanks for the report would make sure to point this python sub-repo to PyPi web-site until we decide what to do. Two repositories of code always lead to problems.

martinayotte, 2017-04-20:
From what I understand, the version from https://pypi.python.org/pypi/pyA20/0.2.1 is only a tarball done from the github version, since pypi.python.org doesn't shows any sources repository.
This probably mean that new 0.2.2 tarball should be done and placed on pypi.python.org.