OLIMEX/OLINUXINO #46: 4.3 LCD is using dithering by choice or accident?

oliv3r, 2017-06-14:
The 4.3 LCD display seems to be a true 24bit screen according to the datasheet (GFT043HA480272Y).

Yet we enable 6 colors per pixel dithering here:
https://github.com/OLIMEX/OLINUXINO/blob/master/SOFTWARE/A20/A20-build-3.4.103/script_change_display_3.4.103/change_display.sh#L556

Why is that? We should be able to set frm to 0 indicating a 24 bit, rather then an 18 bit display and thus disable color dithering.

DanKoloff, 2017-07-07:
Are you sure that changing this parameter does something? My oscilloscope shows some activity of the bits after 18 even if lcd_frm=1... I also can't find any visible difference - I ran two boards with two displays and one of them had lcd_frm=0 while the other lcd_frm=1.

In general, I not also sure if lcd_frm affects the color depth at all. I believe it simply affects the dithering. I also read something similar here, but i'm not sure about the current state: https://lists.denx.de/pipermail/u-boot/2015-January/200035.html

Can you elaborate further on your suggestion? Some tests or proof that lcd_frm actually makes a difference.

net147, 2017-08-07:
Some batches of LCD panels used in LCD-OLinuXino-4.3TS are actually 18-bit and require dithering otherwise you will see banding when displaying grayscale gradients on these displays. The datasheet doesn't necessarily match the panel as Olimex changes their supplier of LCD panels from time to time as availability changes. This means things like viewing angle and 18-bit/24-bit may change but hopefully the physical dimensions and data timings should be compatible as they switch suppliers.

oliv3r, 2017-08-08:
so some displays require dithering, where others do not, but isn't it a bit silly to drive a 24bit screen in 18 bit?

Either sell the display as 18 bit, or have 2 different part numbers I'd say.

oliv3r, 2017-08-08:
@DanKoloff for the fex stuff, i doubt or care it matters, but when driving these displays from mainline, it does matter as you then add it to the device-tree where it is actually picked up afaik.

net147, 2017-08-08:
@oliv3r Mainline doesn't support dithering at the moment. The registers used for dithering are undocumented.

oliv3r, 2017-10-19:
@net147 we'll figure it out eventually assuming the 3.4 registers actually work ...

as for the display, it makes supporting it really hard, if there's 2 kinds, and it's kind of hit-miss which one you get :( i guess 2 options would be needed then, olimex-4.3-24 and olimex-4.3-18 😭 
but I guess it resolves the issue I suppose :)

closed 2017-10-19 by oliv3r