OLIMEX/OLINUXINO #47: Not an eeschema file?

mkschreder, 2017-07-04:
I am not able to open any of the schematics with eeschema v4.0.6. I keep getting error about the file not being a vaild eeschema file. How can I open the schematics with kicad?

TsvetanUsunov, 2017-07-04:
Please upgrade to more decent KiCAD version and will be able to open
The current eschema we use is 4.4.0-83

closed 2017-07-04 by TsvetanUsunov

mkschreder, 2017-07-04:
I'm using latest kicad git trunk. I'm still getting the same error. I'm just opening it with eeschema. 

Error loading schematic file '/data/hardware/olinuxino/HARDWARE/H3-OLinuXino/H3-OLinuXino_Rev_A.sch'.
'/data/hardware/olinuxino/HARDWARE/H3-OLinuXino/H3-OLinuXino_Rev_A.sch' does not appear to be an Eeschema file

mkschreder, 2017-07-04:
Schematic files usually are text files. Are these eagle schematics? I though you guys were using KiCad for all of your projects? Not the case?

TsvetanUsunov, 2017-07-04:
you never mentioned that you open H3 design, yes this is old board made with Eagle

mkschreder, 2017-07-04:
But the only one that is not using eagle in that case is the A64 laptop
design? Do you have kicad BGA footprints and symbols somewhere?

TsvetanUsunov, 2017-07-04:
IIRC we decidedo to move to KiCAD 3 years ago all our new designs which started afterward are with KiCAD, but OLINUXINO project started 2012 and our old designs are with Eagle. Once design is started with Eagle it makes no sense to re-design with new CAD just because you want to add few components in the new revision.

xrbll, 2020-06-09:
so? i have the same error.