OLIMEX/OLINUXINO #49: A64 Mainline - Debian arm64 kernel page state error

erikly01, 2017-12-30:
I was able to get it to boot Debian successfully using the [Sunxi bootloader](https://linux-sunxi.org/Mainline_U-Boot#Boot) and the [Debian ARM64 port](https://wiki.debian.org/Arm64Port), however I get the following kernel errors (below).  

Does anyone have any clues as to how to resolve these problems?

```
[    1.186363] Freeing unused kernel memory: 448K
[    1.892064] systemd[1]: System time before build time, advancing clock.
[    1.973534] x_tables: version magic '4.13.13 SMP preempt mod_unload modversions aarch64' should be '4.13.13 SMP mod_unload aarch64'
[    1.985513] systemd[1]: Failed to insert module 'ip_tables': Exec format error
[    2.031000] systemd[1]: systemd 236 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN2 +IDN default-hierarchy=hybrid)
[    2.052054] systemd[1]: Detected architecture arm64.

Welcome to Debian GNU/Linux buster/sid!

[    2.069222] systemd[1]: Set hostname to <olimex>.
[    2.274027] BUG: Bad page state in process systemd-cryptse  pfn:7dd29UG: Bad page state in process systemd-cryptse  pfn:7dd29
[    2.280474] page:ffffffbf00f74a40 count:1 mapcount:0 mapping:          (null) index:0x1
[    2.288468] flags: 0x0()
[    2.291003] raw: 0000000000000000 0000000000000000 0000000000000001 00000001ffffffff
[    2.298735] raw: dead000000000100 dead000000000200 0000000000000000 0000000000000000
[    2.306463] page dumped because: nonzero _refcount
[    2.311244] Modules linked in:
[    2.314300] CPU: 0 PID: 119 Comm: systemd-cryptse Not tainted 4.13.13 #13
[    2.321076] Hardware name: Olimex A64-Olinuxino (DT)
[    2.326031] Call trace:
[    2.328485] [<ffffff8008086f68>] dump_backtrace+0x0/0x210
[    2.333879] [<ffffff8008087254>] show_stack+0x14/0x20
[    2.338926] [<ffffff80085c4360>] dump_stack+0x98/0xb8
[    2.343972] [<ffffff800813fc30>] bad_page+0xe0/0x140
[    2.348931] [<ffffff800813fce4>] free_pages_check_bad+0x54/0xb8
[    2.354842] [<ffffff8008140008>] free_pcppages_bulk+0x218/0x2f0
[    2.360754] [<ffffff8008141bf0>] free_hot_cold_page+0x1a8/0x1f0
[    2.366665] [<ffffff8008141f54>] free_hot_cold_page_list+0x34/0x60
[    2.372837] [<ffffff800814939c>] release_pages+0x27c/0x338
[    2.378319] [<ffffff800817ab18>] free_pages_and_swap_cache+0xb0/0xc8
[    2.384666] [<ffffff8008164c00>] tlb_flush_mmu_free+0x40/0x60
[    2.390404] [<ffffff8008165a98>] arch_tlb_finish_mmu+0x48/0x78
[    2.396229] [<ffffff8008165cb4>] tlb_finish_mmu+0x24/0x50
[    2.401622] [<ffffff800816dbf0>] exit_mmap+0xb0/0xf8
[    2.406582] [<ffffff800809f2ec>] mmput+0x44/0xf8
[    2.411194] [<ffffff80080a522c>] do_exit+0x26c/0x990
[    2.416152] [<ffffff80080a59b4>] do_group_exit+0x34/0xa0
[    2.421456] [<ffffff80080a5a30>] __wake_up_parent+0x0/0x28
[    2.426934] [<ffffff8008082ca0>] __sys_trace_return+0x0/0x4
[    2.432496] Disabling lock debugging due to kernel taint
```

jcstaudt, 2018-08-02:
Cross-referencing the [Olimex forum thread](https://www.olimex.com/forum/index.php?topic=6003.0).

Any progress figuring this out?  I have the same goal: getting Debian with a mainline kernel + mainline u-boot to run on the A64 OLinuXino.  Have you found any other good references from Sunxi or Debian?

hehopmajieh, 2018-08-02:
Please attach kernel config file and kernel version

closed 2020-10-13 by DanKoloff

FreshLuoBoGan, 2022-07-01:
i have this problem,how to resolve it
[image]