OLIMEX/OLINUXINO #52: Latest ubuntu a64 image torrent hangs on bootloader

Lewiscowles1986, 2018-02-13:
https://www.olimex.com/wiki/images/1/1b/A64olinuxino_ubuntu_16.04.3_20171110.torrent

flashed using etcher (so verified write), retried using several microSD cards. Unsure what the issue is.

https://www.olimex.com/wiki/images/d/d5/A64-ubuntu_built_preliminary_rel_2.torrent (which looks like the image in this git repo) works

https://www.olimex.com/Products/OLinuXino/A64/A64-OLinuXino/open-source-hardware

Using the 1GB a64 board with onboard storage

DanKoloff, 2018-02-14:
Thanks for the report. The image seems to boot fine here but the video output is not set to HDMI. I also used Ethcher to prepare the card. I've added a note about the HDMI in the web-page of the board. The default output would be fixed in next official release.

I believe that you might have connected the USB TTL wrongfully. Or the different images might bring different power supply requirements.

Lewiscowles1986, 2018-02-14:
Thanks. It was a 4-pin USB->TTL. 

I also use for a number of other SBC's but perhaps I need to re-read the docs, I remember something specific about 3.3v.

If it's fine by you I'd sooner leave this open until next release, that way you don't suffer duplicates until that happens. 

Thanks again!

jcstaudt, 2018-08-14:
Hi @Lewiscowles1986, did you have any luck with eMMC-booting [Ubuntu 16.04.3 20180523 rel5](https://www.olimex.com/wiki/images/4/43/A64olinuxino_ubuntu_16.04.3_20180523_rel5.torrent)?

I only have tested with SD-boot, but this image works with UART and HDMI, though it doesn't always boot up in the most stable manner and sometimes requires a hard reset.  Most of the time the errors preventing boot are related to not discovering a USB hub, etc.  I was able to perform a distribution upgrade to 18.04LTS without issue (but only when I did it straight away from a fresh image - not after installing several software packages).

Lewiscowles1986, 2018-08-14:
Hi @jcstaudt,
Putting it on my list to try again at some point this week. TBF it's been sitting on the side since the end of the week I ran this. 

I think I ran armbian on it, but stability was something a bit meh.

jcstaudt, 2018-08-14:
I'm attempting to get Armbian running properly when I have some spare time.  Here's the related [Armbian thread with an update from Igor Pečovnik](https://forum.armbian.com/topic/5862-a64-olinuxino).

closed 2020-10-13 by DanKoloff