OLIMEX/OLINUXINO #61: USB ESD protection?

mkschreder, 2018-03-20:
Hello,

Is there no USB ESD protection on any of the olinuxino boards? I have looked at a few schematics and I could not find it. Is it usually built into the Allwinner soc?

DanKoloff, 2018-03-21:
USB cables and connector are grounded and this eliminates some of the ESD. There is no separate ESD protection for the USB ports. Additional IC or Zener diodes can be placed if somebody frequently has such problems. The main reason for the lack of such protection is that we haven't encountered ESD problems with the USBs and complicating the schematics further sometimes leads to more problems.

mkschreder, 2018-03-21:
Kinda hard to place extra diodes if it is a full board and not a som that plugs into a custom carrier. I also second the point about EC compliance. Good news is we can always just redesign the schematic and make our own versions. Thanks!

DanKoloff, 2018-03-21:
We have Allwinner SOM boards and you can use the open design of the evaluation boards compatible with the SOMs to tailor the USB circuit as you might like:

A20-SOM: https://www.olimex.com/Products/SOM/A20/A20-SOM/
A20-SOM204: https://www.olimex.com/Products/SOM204/A20/A20-SOM204/
A13-SOM-256: https://www.olimex.com/Products/SOM/A13/A13-SOM-256/
A13-SOM-512: https://www.olimex.com/Products/SOM/A13/A13-SOM-512/

closed 2018-03-21 by mkschreder