OLIMEX/OLINUXINO #64: Eagle File unable to load in kicad, easyeda, openboardview

Nathan-Jonson, 2018-06-02:
I am unable to load the brd file in any program that I can find. Is it possible that the format is old? 
The error message that comes up in both kicad is ``` XML parsing error: 'not well formed (Invalid token)' at line one.```
The easyeda error message is something about an invalid utf character at line 1. 
openboardview.org has same issue. 

Any help appreciated.

Nathan-Jonson, 2018-06-02:
I looked into it a little more and it appears that the file is binary, not XML. Would it be possible to add XML format eagle files?

DanKoloff, 2018-06-04:
The design was made with CadSoft's EAGLE (now owned by Autodesk).

Nathan-Jonson, 2018-06-04:
Yes, I know. somewhere arround version 6 or so, they changed their file format from a binary to an xml format. A lot of tools can open the new xml format, however, only eagle can open the old binary files. My solution was to poen them in a free trial version of eagle, then save them in the new format.

easyw, 2018-06-04:
@DanKoloff 
the solution, as pointed out by @Nathan-Jonson is to upload the file @GitHub in XML format instead of binary.
This can be done within Eagle, changing the way to save the file.

TsvetanUsunov, 2018-06-04:
@Nathan-Jonson the Eagle version we have full licensee and used to make this board is 4.16 we never upgraded since this version and this is the only file format it can save. To the best of my knowledge later versions read this format just fine but we do not use Eagle since 2 years so things may have changed.