OLIMEX/OLINUXINO #68: What distro is best to use for the latest A20-build?

jcalfee, 2018-12-06:
I'm following this guide:

https://raw.githubusercontent.com/OLIMEX/OLINUXINO/master/SOFTWARE/A20/A20-build-3.4.103-release-8/BUILD_DESCRIPTION_A20_Olimex_kernel_3.4.103%2B_Jessie_rel_8.txt

These packages are not in apt-get in debian stretch (my typical selection) or debian squeeze: 
`gcc-4.7-arm-linux-gnueabihf ncurses-dev`

What distro do you recommend for the build environment?  I'm using Docker so it is not a big deal to change it.  It is just tricky to find the distro that will both find the old version of gcc (gcc-4.7 gnueabihf) even when I try adding this repository:
```bash
deb http://www.emdebian.org/debian squeeze main
in the file below:
/etc/apt/sources.list
```

Please consider using something like docker, it will help to re-create the environment and builds..

DanKoloff, 2018-12-07:
Thank you for the contribution. Our sunxi kernel images and instructions were made using Ubuntu 16.04.3 LTS machine so it is entirely possible to have differences and problems with different distributions.

We switched to Ubuntu 18.04.1 LTS for the building of our Armbian-based images. Expect them out soon.

jcalfee, 2018-12-07:
Ok, next time I'll try Ubuntu 16 and look for notes about the upgrade.  Thanks..

closed 2018-12-07 by jcalfee