OLIMEX/OLINUXINO #69: Question: step 5 A20-build-3.4.103-release-8

jcalfee, 2018-12-13:
Step 5 says to write Uboot and **u-boot-sunxi-with-spl.bin** to the SD card.  I am however working with the LIME2 (not the LIME) so I did not build **u-boot-sunxi-with-spl.bin**.  This file is build in the `2.4 A20-OLinuXino-LIME board` section.  I ran the make in `2.1 A20-OLinuXino-LIME2 board` instead.

https://raw.githubusercontent.com/OLIMEX/OLINUXINO/master/SOFTWARE/A20/A20-build-3.4.103-release-8/BUILD_DESCRIPTION_A20_Olimex_kernel_3.4.103%2B_Jessie_rel_8.txt

Should I run the LIME build too?

jcalfee, 2018-12-13:
Also see 7.6, `# cp uEnv.txt.bin /mnt/sd` should be `# cp uEnv.txt /mnt/sd`

jcalfee, 2018-12-13:
Also, no seeds for `Debian_FS_34_103_Jessy_8.1_A20-olimex_rel_4.torrent`..

jcalfee, 2018-12-13:
Section 8: Change `# umount /mnt/sdX2` to `# umount /dev/sdX2`

DanKoloff, 2018-12-13:
> Should I run the LIME build too?

No, don't. LIME2 and LIME boards have different defconfigs - A20-OLinuXino-Lime2_defconfig and A20-OLinuXino-Lime_defconfig. I think that you missed the separator below the LIME (2.4) section. It is important to check if everything went ok with this part "At the end of the process you can check if everything is OK by using:"

> Also see 7.6, # cp uEnv.txt.bin /mnt/sd should be # cp uEnv.txt /mnt/sd

Corrected.

> Section 8: Change # umount /mnt/sdX2 to # umount /dev/sdX2

Corrected.

Thanks for the feedback.

jcalfee, 2018-12-13:
>think that you missed the separator below the LIME (2.4) section

Correct.

Can you please tell me what this is?
`./scripts/dtc-version.sh: line 18: dtc: command not found`

I don't think it is mail or xen as found here:
https://packages.debian.org/search?keywords=dtc&searchon=names&suite=stretch&section=all

pfalcon, 2018-12-13:
Google for "device tree compiler".

jcalfee, 2018-12-13:
I understand .. I don't know why there is no make command for dtc or why I did not end up with this made already by following the directions.

jcalfee, 2018-12-13:
ok, I found it in the debian apt package manager, but an older version..

jcalfee, 2018-12-13:
This is getting messy.   I think I'm going to start over with Ubuntu 16.04.3 LTS  ..

closed 2018-12-13 by jcalfee

jcalfee, 2018-12-17:
Move the CD command up before the `git reset ...` and after `git clone ...`

```bash
Move the CD 

Chapter 3. Building sunxi kernel from sources for A20 boards

---

The sunxi kernel sources for A20 are available at GitHub. Note that the our build uses this exact commit: 

9a1cd034181af628d4145202289e1993c1687db6 from Sat Oct 4 07:59:35 2014 +0300

You can download the latest kernel sources using the following command:

# git clone https://github.com/linux-sunxi/linux-sunxi

You can switch to this exact commit with the next command:

# git reset --hard 9a1cd034181af628d4145202289e1993c1687db6

After the download navigate to the kernel directory:

# cd linux-sunxi/
```