OLIMEX/OLINUXINO #75: Question about power supply using AXP209 ic

shakenov-chinga, 2019-02-27:
Hi, everybody!
I am working on my personal project using allwinner A13 SoC. Thank you very much for your work and for sharing the schematics!
Right now I'm trying to use AXP209 PMIC, and I have a question about its usage:
I have noticed that wherever you use AXP209, you also use other parts, like SY8009AAAC and MP1584EN (or MP1482DS) and others. But in some cases, like in A13-SOM512 version you use only SY8009AAAC without AXP209, and also in A10S-OLinuXino-MICRO use only AXP152 without any additional parts. As I can deduce, one reason for this is that you don't have battary in these cases. My question is: why can't we use AXP209 without additional parts? As much as I can understand from AXP209 datasheet, they are not so necessary.
Thanks in advance!

shakenov-chinga, 2019-02-27:
Also found out that typical application schematics for AXP223 and AXP221 in datasheet also do not require additional IC parts, and you don't use additional parts with them too. Can't we use AXP209 the same way?

closed 2020-10-13 by DanKoloff

diogoviannaaraujo, 2023-02-24:
Wonder what were your experiences with the AXP209 IC.