OLIMEX/OLINUXINO #76: RT5350 release 3

ec1oud, 2019-08-17:
https://www.olimex.com/wiki/RT5350F-OLinuXino says release 3 was made in April.  Where is it available?

DanKoloff, 2019-08-19:
It can be found here: ftp://staging.olimex.com/Ralink/RT5350F-OLinuXino/1.latest_images/

DanKoloff, 2019-08-20:
The repository contains 6 files, 2 for each variant of Olimex-made RT5350 boards. We have three variants of RT5350F-OLinuXino - RT5350F-OLinuXino, RT5350F-OLinuXino-EVB, and RT5350F-OLinuXino-EVB+ (which is latest revisions of RT5350F-OLinuXino-EVB, starting revision C, with SD card and reset button, among others).  One of the files is for u-boot upload, the other is for luci upload. Read more about the upgrading here: https://openwrt.org/docs/guide-user/installation/generic.sysupgrade

ec1oud, 2019-08-21:
Thanks.  You might want to update the web page to link to that.

I'm running an 18.06.1 upstream release, which seems to work.  Are you adding more features to your own builds, or upstreaming everything to openwrt?  For example when it comes to gpio, I can control the relays and read the button state, but haven't been able to figure out how to wiggle other pins.  I want to try to use one of the pins on the UEXT connector as a chip select (for a CS5490).  I guess that means editing the dts file to make it available for that, if it's not already.  I would ask on the forum, but I've never been able to log in there (it's been more than a year); it keeps saying I have to register my account, so it will send an email to do that, but the email never comes.

DanKoloff, 2019-09-17:
Not huge differences compared to 18.06, visible here:

https://github.com/OLIMEX/openwrt/commit/16d698d67bf56af0a0370608f20f625a5002f07e

closed 2020-10-13 by DanKoloff