OLIMEX/OLINUXINO #78: CONFIG_SQUASHFS_XZ

cyberb, 2019-09-08:
Is it possible to include this flag to get snapd support on Lime2 board?

````
CONFIG_SQUASHFS_XZ=y
````

I am using this image for syncloud.org project:
https://www.olimex.com/wiki/A20-OLinuXino-LIME2#Linux_images

DanKoloff, 2019-09-09:
What image exactly are you using and where did you get it from? Did you try the Armbian images, it seems to be enabled in them:

ftp://staging.olimex.com/Allwinner_Images/A20-OLinuXino/1.latest_mainline_images/bionic/images/

https://www.olimex.com/wiki/ArmbianHowTo

cyberb, 2019-09-09:
I was trying this one:
https://www.olimex.com/wiki/images/2/24/A20-lime2_mainline_uboot_sunxi_kernel_3.4.103_jessie_NAND_rel_13.torrent

Ok, let me test armbian image.

closed 2020-10-13 by DanKoloff