OLIMEX/OLINUXINO #82: please mention release date for each revision

jonassmedegaard, 2020-03-09:
README.md files documenting board revisions - e.g. `HARDWARE/A64-OLinuXino/README.md` would be more helpful if the date of release was included.

With release date, it would be possibly to rule out newer changes if I didn't know the revision of a board but knew when it was bought (if bought December 2018 then I can certainly rule out any board changes made after that time).

TsvetanUsunov, 2020-03-09:
define date of release :)

we first design the new revision, then make prototypes, then test them, if everything is OK run production of PCB, then if there are old revision blank PCBs we first use them all then replace the PCBs, it's long process. even when we start selling there are older revisions which are at our distributors on stock which may be confusing for some customers i.e. they see new PCB revision on GitHub but buy older revision from distributor.

jonassmedegaard, 2020-03-09:
the first ever date that anyone outside the company could get that revision of the board through normal channels (sales, gifts, etc.).

(i.e. if a board revision "leaks" - accidentally or intentionally - then I'd classify that as some kind of "pre-release" of that revision board)

I understand that multiple revisions exist concurrently "in the wild".  That's why my example was only about how I could eliminate _newer_ board revisions, not older ones.