OLIMEX/OLINUXINO #87: UBOOT ISSUE with ethernet GMAC A20

akshar001, 2021-03-19:
We have used the hardware reference with KSZ9031 phy chip,

The problem is when in U-BOOT we try the command dhcp it shows DMA RESET TIMEOUT,

when we write mii dump 1 1 command ( our phy device is id is - 1 ) it actually shows the mii interface and link status is also 1 when ethernet cable is connected.

We think problem is in GMAC is not being identified in U-BOOT. Can anyone help with us?

DanKoloff, 2021-04-02:
Hmm, are you are creating your own board based on one of the Olimex designs?

What design is your board based on?

akshar001, 2021-04-02:
@DanKoloff it's fixed. The official image didn't included GMAC but letter we changed u-boot kernel and it's working.
We used lime2 design.

closed 2021-04-02 by akshar001