OLIMEX/OLINUXINO #88: A20-OLinuXino Linux uses kernel default qdisc

androidcho, 2021-05-28:
After some research I have found that the kernel's default "qdisc pfifo_fast" is used (or more likely not set).
It uses it for both the ETH interface of A20-LIME2 and the SocketCAN interface.
Setting "qdisc fq_codel" improved the ethernet link quality alot.
Setting "qdisc noqueue" for the CAN interface stopped it losing CAN frames.

DanKoloff, 2021-05-28:
Which LInux image did you use? One of the Olimage images?

androidcho, 2021-05-28:
> Which Linux image did you use? One of the Olimage images?
To be precise:
A20-OLinuXino-buster-minimal-20210318-122357.img

DanKoloff, 2021-05-28:
Perfect, thanks for this info and sharing your experience and advice. I will point the developers to this note to look into it.

closed 2026-10-27 by DanKoloff