OLIMEX/OLINUXINO #90: Where is the source for A13 script.bin?

thom-nic, 2022-04-14:
I'm following the documentation in the [A13-SOM-build directory](https://github.com/OLIMEX/OLINUXINO/blob/master/SOFTWARE/A13/A13-SOM-build/A13_SOM_kernel_3.4.90_building_description_with_mainline_u-boot_2.txt) and intend to build a custom image with mainline u-boot and kernel. The `script.bin` is download ed from a Google Drive folder. Could you please provide source for this file and instructions to build it? (is it just built with mkimage?)

DanKoloff, 2022-04-15:
Script.bin can be transformed to script.fex via a tool, and the script.fex is a text file (can be checked or edited via any text editor). It is explained here:

https://www.olimex.com/wiki/How_to_edit_board_configurations_and_definitions_in_the_official_Debian_Linux

DanKoloff, 2022-04-15:
Notice that we recommend using Olimage Linux images:

https://github.com/OLIMEX/OLINUXINO/blob/master/DOCUMENTS/OLIMAGE/Olimage-guide.pdf

closed 2026-09-21 by DanKoloff