OLIMEX/OLINUXINO #91: Issue in compilation of Olimex A20-build-3.4.103-release-8 image

harshalbhavsar632, 2022-09-16:
Hello Team,

We are trying to build "A20-build-3.4.103-release-8" this version on Ubuntu-20.04. We are referring "https://github.com/OLIMEX/OLINUXINO/tree/master/SOFTWARE/A20/A20-build-3.4.103-release-8" this link to build the image. But unfortunately we are facing issue to build the Kernel Image. I am providing some system details as well as error logs.

As per my understanding this build required GCC-4.7 version but I am not able to get this version any where and we required this Olimex version only. Please suggest us suitable changes.

I also tried to install Ubuntu 12.04 LTS but as we know its not supported any more.

Extra Changes which we did:
As per Document : apt-get install gcc-4.7-arm-linux-gnueabihf ncurses-dev uboot-mkimage build-essential git

We executed : apt-get install gcc-arm-linux-gnueabihf libncurses-dev u-boot-tools build-essential git

----------------------------------------------

System Details:
5.15.0-46-generic #49~20.04.1-Ubuntu SMP Thu Aug 4 19:15:44 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

-----------------------------------------------------------------
Error Logs:

a20-olimex/linux-sunxi# make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -j4 uImage
scripts/kconfig/conf --silentoldconfig Kconfig
warning: (VIDEO_PVRUSB2_DVB && VIDEO_CX18 && VIDEO_CX23885 && VIDEO_CX88_DVB && VIDEO_SAA7164 && DVB_USB_DIB0700) selects DVB_S5H1411 which has unmet direct dependencies (MEDIA_SUPPORT && DVB_CAPTURE_DRIVERS && DVB_CORE && I2C)
warning: (VIDEO_PVRUSB2_DVB && VIDEO_CX18 && VIDEO_CX23885) selects DVB_S5H1409 which has unmet direct dependencies (MEDIA_SUPPORT && DVB_CAPTURE_DRIVERS && DVB_CORE && I2C)
warning: (VIDEO_PVRUSB2_DVB && VIDEO_EM28XX_DVB && VIDEO_CX23885 && VIDEO_CX88_DVB && DVB_USB_CXUSB) selects DVB_LGDT330X which has unmet direct dependencies (MEDIA_SUPPORT && DVB_CAPTURE_DRIVERS && DVB_CORE && I2C)
warning: (VIDEO_PVRUSB2_DVB && VIDEO_CX23885 && VIDEO_SAA7134_DVB && VIDEO_SAA7164) selects DVB_TDA10048 which has unmet direct dependencies (MEDIA_SUPPORT && DVB_CAPTURE_DRIVERS && DVB_CORE && I2C)
warning: (VIDEO_PVRUSB2_DVB && VIDEO_CX18 && VIDEO_CX23885 && VIDEO_CX88_DVB && VIDEO_SAA7164 && DVB_USB_DIB0700) selects DVB_S5H1411 which has unmet direct dependencies (MEDIA_SUPPORT && DVB_CAPTURE_DRIVERS && DVB_CORE && I2C)
warning: (VIDEO_PVRUSB2_DVB && VIDEO_CX18 && VIDEO_CX23885) selects DVB_S5H1409 which has unmet direct dependencies (MEDIA_SUPPORT && DVB_CAPTURE_DRIVERS && DVB_CORE && I2C)
warning: (VIDEO_PVRUSB2_DVB && VIDEO_EM28XX_DVB && VIDEO_CX23885 && VIDEO_CX88_DVB && DVB_USB_CXUSB) selects DVB_LGDT330X which has unmet direct dependencies (MEDIA_SUPPORT && DVB_CAPTURE_DRIVERS && DVB_CORE && I2C)
warning: (VIDEO_PVRUSB2_DVB && VIDEO_CX23885 && VIDEO_SAA7134_DVB && VIDEO_SAA7164) selects DVB_TDA10048 which has unmet direct dependencies (MEDIA_SUPPORT && DVB_CAPTURE_DRIVERS && DVB_CORE && I2C)
  CHK     include/linux/version.h
  UPD     include/linux/version.h
  HOSTCC  scripts/genksyms/genksyms.o
  CC      scripts/mod/empty.o
  WRAP    arch/arm/include/generated/asm/auxvec.h
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/percpu.h
  WRAP    arch/arm/include/generated/asm/poll.h
  WRAP    arch/arm/include/generated/asm/resource.h
  WRAP    arch/arm/include/generated/asm/sections.h
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  HOSTCC  scripts/mod/mk_elfconfig
  MKELF   scripts/mod/elfconfig.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTCC  scripts/selinux/genheaders/genheaders
  SHIPPED scripts/genksyms/lex.lex.c
  SHIPPED scripts/genksyms/keywords.hash.c
  SHIPPED scripts/genksyms/parse.tab.h
  SHIPPED scripts/genksyms/parse.tab.c
  HOSTCC  scripts/selinux/mdp/mdp
  HOSTCC  scripts/genksyms/lex.lex.o
  HOSTCC  scripts/genksyms/parse.tab.o
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/kallsyms
  HOSTLD  scripts/genksyms/genksyms
  HOSTCC  scripts/mod/sumversion.o
  HOSTCC  scripts/pnmtologo
  HOSTCC  scripts/conmakehash
  HOSTCC  scripts/bin2c
  HOSTLD  scripts/mod/modpost
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
In file included from include/linux/compiler.h:48,
                 from include/linux/stddef.h:4,
                 from include/linux/posix_types.h:4,
                 from include/linux/types.h:17,
                 from include/linux/page-flags.h:8,
                 from kernel/bounds.c:9:
include/linux/compiler-gcc.h:103:1: fatal error: linux/compiler-gcc9.h: No such file or directory
  103 | #include gcc_header(__GNUC__)
      | ^~~~
compilation terminated.
make[1]: *** [/home/harshal/a20-olimex/linux-sunxi/./Kbuild:36: kernel/bounds.s] Error 1
make: *** [Makefile:987: prepare0] Error 2
-----------------------------------------

GCC Version:
a20-olimex/linux-sunxi# gcc --version
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Thanks,
Harshal

hehopmajieh, 2022-09-16:
Hi,
GCC 4.7 toolchain can be found here :
https://releases.linaro.org/archive/13.05/components/toolchain/gcc-linaro/4.7/

Do not forget to set $PATH to no toolchain files location
p.s
https://askubuntu.com/questions/113291/how-do-i-install-gcc-4-7
~Best

harshalbhavsar632, 2022-09-17:
Thanks for your support.

I am trying to install your provided GCC compiler. But I am facing compile time error for it.

Error Logs:
./../.././libjava/prims.cc: In function ‘void _Jv_catch_fpe(int, siginfo_t*, void*)’:
../../.././libjava/prims.cc:192:3: error: invalid use of incomplete type ‘struct _Jv_catch_fpe(int, siginfo_t*, void*)::ucontext’
../../.././libjava/prims.cc:192:3: error: forward declaration of ‘struct _Jv_catch_fpe(int, siginfo_t*, void*)::ucontext’
make[5]: *** [Makefile:9895: prims.lo] Error 1
make[5]: Leaving directory '/home/harshal/Downloads/gcc-linaro-4.7-2013.05/x86_64-unknown-linux-gnu/32/libjava'
make[4]: *** [Makefile:10204: all-recursive] Error 1
make[4]: Leaving directory '/home/harshal/Downloads/gcc-linaro-4.7-2013.05/x86_64-unknown-linux-gnu/32/libjava'
make[3]: *** [Makefile:12760: multi-do] Error 1
make[3]: Leaving directory '/home/harshal/Downloads/gcc-linaro-4.7-2013.05/x86_64-unknown-linux-gnu/libjava'
make[2]: *** [Makefile:12724: all-multi] Error 2
make[2]: Leaving directory '/home/harshal/Downloads/gcc-linaro-4.7-2013.05/x86_64-unknown-linux-gnu/libjava'
make[1]: *** [Makefile:18522: all-target-libjava] Error 2
make[1]: Leaving directory '/home/harshal/Downloads/gcc-linaro-4.7-2013.05'
make: *** [Makefile:906: all] Error 2

---------------------------------------------

I have tried : 
https://www.mail-archive.com/gcc-patches@gcc.gnu.org/msg168317.html

But still facing same issue. Do you have any reference.

Thanks,
Harshal

harshalbhavsar632, 2022-09-17:
Thanks for your support.

We are successfully installed GCC-4.7. But still facing same issue.

Current System Information:

root@harshal-Latitude-E5430-vPro:/usr/bin# ls -lrt gcc*
lrwxrwxrwx 1 root root      12 Mar 20  2020 gcc-ranlib -> gcc-ranlib-9
lrwxrwxrwx 1 root root       8 Mar 20  2020 gcc-nm -> gcc-nm-9
lrwxrwxrwx 1 root root       8 Mar 20  2020 gcc-ar -> gcc-ar-9
lrwxrwxrwx 1 root root      29 Mar  9  2022 gcc-ranlib-9 -> x86_64-linux-gnu-gcc-ranlib-9
lrwxrwxrwx 1 root root      25 Mar  9  2022 gcc-nm-9 -> x86_64-linux-gnu-gcc-nm-9
lrwxrwxrwx 1 root root      25 Mar  9  2022 gcc-ar-9 -> x86_64-linux-gnu-gcc-ar-9
lrwxrwxrwx 1 root root      22 Mar  9  2022 gcc-9 -> x86_64-linux-gnu-gcc-9
-rwxr-xr-x 1 root root 2133208 Sep 17 19:12 gcc-4.7
lrwxrwxrwx 1 root root       7 Sep 17 19:13 gcc -> gcc-4.7

------------------------
GCC Version Command Output:
root@harshal-Latitude-E5430-vPro:/usr/bin# gcc --version
gcc (Linaro GCC 4.7-2013.05) 4.7.4 20130508 (prerelease)
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

------------------------------------

Error Logs:
 HOSTCC  scripts/selinux/genheaders/genheaders
In file included from include/linux/compiler.h:48,
                 from include/linux/stddef.h:4,
                 from include/linux/posix_types.h:4,
                 from include/linux/types.h:17,
                 from include/linux/page-flags.h:8,
                 from kernel/bounds.c:9:
include/linux/compiler-gcc.h:103:1: fatal error: linux/compiler-gcc9.h: No such file or directory
  103 | #include gcc_header(__GNUC__)
      | ^~~~
compilation terminated.
make[1]: *** [/home/harshal/a20-olimex/linux-sunxi/./Kbuild:36: kernel/bounds.s] Error 1
make: *** [Makefile:987: prepare0] Error 2
make: *** Waiting for unfinished jobs....

--------------------------------------

Any Suggestion..

harshalbhavsar632, 2022-09-17:
Hello,

We resolved this issue using "ln -s compiler-gcc4.h compiler-gcc9.h" But now facing issue in return_address.c

Error Logs:
rch/arm/kernel/return_address.c:65:7: error: redefinition of ‘return_address’
   65 | void *return_address(unsigned int level)
      |       ^~~~~~~~~~~~~~
In file included from include/linux/ftrace.h:19,
                 from arch/arm/kernel/return_address.c:12:
/home/harshal/a20-olimex/linux-sunxi/arch/arm/include/asm/ftrace.h:48:21: note: previous definition of ‘return_address’ was here
   48 | extern inline void *return_address(unsigned int level)
      |                     ^~~~~~~~~~~~~~
make[1]: *** [scripts/Makefile.build:307: arch/arm/kernel/return_address.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:947: arch/arm/kernel] Error 2

----------------------------------------

We tried following points but still not resolved:
Reference : https://github.com/gumstix/live-build/issues/5

root@harshal-Latitude-E5430-vPro:/usr/bin# ls -lrt gcc*
lrwxrwxrwx 1 root root      29 Mar  9  2022 gcc-ranlib-9 -> x86_64-linux-gnu-gcc-ranlib-9
lrwxrwxrwx 1 root root      25 Mar  9  2022 gcc-nm-9 -> x86_64-linux-gnu-gcc-nm-9
lrwxrwxrwx 1 root root      25 Mar  9  2022 gcc-ar-9 -> x86_64-linux-gnu-gcc-ar-9
lrwxrwxrwx 1 root root      22 Mar  9  2022 gcc-9 -> x86_64-linux-gnu-gcc-9
-rwxr-xr-x 1 root root 2133208 Sep 17 19:12 gcc-4.7
lrwxrwxrwx 1 root root       7 Sep 17 19:13 gcc -> gcc-4.7
-rwxr-xr-x 1 root root  118608 Sep 17 20:46 gcc-ar-4.7
-rwxr-xr-x 1 root root  118536 Sep 17 20:46 gcc-nm-4.7
-rwxr-xr-x 1 root root  118552 Sep 17 20:47 gcc-ranlib-4.7
lrwxrwxrwx 1 root root      10 Sep 17 20:49 gcc-ar -> gcc-ar-4.7
lrwxrwxrwx 1 root root      14 Sep 17 20:49 gcc-ranlib -> gcc-ranlib-4.7
lrwxrwxrwx 1 root root      10 Sep 17 20:49 gcc-nm -> gcc-nm-4.7

We had soft link all GCC-4.7 compiled binary but still same error. 

Please let me know if suggestion.

Thanks,
Harshal

harshalbhavsar632, 2022-09-24:
Hello Team,

We want to compile kernel for A20-OLinuXino-MICRO. Can anyone please suggest any release. Which is compiling properly on Ubuntu-20.04 version or any latest release.

Thanks,
Harshal