OLIMEX/OLINUXINO #93: Unable to reboot

stepfr64, 2023-01-10:
Hello,
Using Lime2 revG2 hardware, i tested nearly all reboot commands (reboot, poweroff --reboot, systemctl start reboot.target...). I also tried to change RebootWatchdogSec to 8s to use watchdog to help rebooting. But no luck, the system just hang (red led is on near power, but everything else is switch off).

Any way to reboot this arm device (when away) ? I can not remove the power cord remotely !

Thanks

Step

stepfr64, 2023-01-10:
the end of the log 👍 
Jan 10 01:48:00 alarm systemd-journald[207]: Journal stopped
Jan 10 01:48:00 alarm systemd-journald[207]: Received SIGTERM from PID 1 (systemd-shutdow).
Jan 10 01:48:00 alarm systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Jan 10 01:48:00 alarm systemd-shutdown[1]: Syncing filesystems and block devices.
Jan 10 01:48:00 alarm systemd-shutdown[1]: Watchdog running with a timeout of 8s.
Jan 10 01:47:59 alarm systemd-shutdown[1]: Using hardware watchdog 'sunxi-wdt', version 0, device /dev/watchdog0
Jan 10 01:47:59 alarm kernel: watchdog: watchdog0: watchdog did not stop!
Jan 10 01:47:59 alarm systemd[1]: Watchdog running with a timeout of 8s.
Jan 10 01:47:59 alarm systemd[1]: Using hardware watchdog 'sunxi-wdt', version 0, device /dev/watchdog0
Jan 10 01:47:59 alarm audit: BPF prog-id=0 op=UNLOAD
Jan 10 01:47:59 alarm audit: BPF prog-id=0 op=UNLOAD
Jan 10 01:47:59 alarm audit: BPF prog-id=0 op=UNLOAD
Jan 10 01:47:59 alarm audit: BPF prog-id=0 op=UNLOAD
Jan 10 01:47:59 alarm audit: BPF prog-id=0 op=UNLOAD
Jan 10 01:47:59 alarm audit: BPF prog-id=0 op=UNLOAD
Jan 10 01:47:59 alarm audit: BPF prog-id=0 op=UNLOAD
Jan 10 01:47:59 alarm systemd[1]: Shutting down.
Jan 10 01:47:59 alarm systemd[1]: Reached target System Reboot.
Jan 10 01:47:59 alarm audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=_ msg='unit=systemd-reboot comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 10 01:47:59 alarm audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=_ msg='unit=systemd-reboot comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 10 01:47:59 alarm systemd[1]: Finished System Reboot.
Jan 10 01:47:59 alarm systemd[1]: systemd-reboot.service: Deactivated successfully.
Jan 10 01:47:59 alarm systemd[1]: Reached target Late Shutdown Services.
Jan 10 01:47:59 alarm systemd[1]: Reached target System Shutdown.
Jan 10 01:47:59 alarm audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=_ msg='unit=systemd-tmpfiles-setup-dev comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 10 01:47:59 alarm systemd[1]: Stopped Create Static Device Nodes in /dev.
Jan 10 01:47:59 alarm systemd[1]: systemd-tmpfiles-setup-dev.service: Deactivated successfully.
Jan 10 01:47:59 alarm audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=_ msg='unit=systemd-remount-fs comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 10 01:47:59 alarm systemd[1]: Stopped Remount Root and Kernel File Systems.
Jan 10 01:47:59 alarm systemd[1]: systemd-remount-fs.service: Deactivated successfully.
Jan 10 01:47:59 alarm systemd[1]: Reached target Unmount All Filesystems.

stepfr64, 2023-01-11:
I use last Archlinuxarm image for lime2. Seems to reboot using armbian ( I tried) . So it must be a software image issue. I opened a ticket on archlinux forum also. If you already know this problem, i will forward any information

DanKoloff, 2023-01-12:
I also tested with Olimage Linux and it seems to reboot fine. It is definitely a software issue. Usually reboots and shutdowns are related to driver handling power management unit, in this case AXP209. Anyway, I am closing this issue since it is not something that Olimex team can fix. Maybe give Olimage Linux a try if you haven't already, you can find info about it here:

https://github.com/OLIMEX/OLINUXINO/blob/master/DOCUMENTS/OLIMAGE/Olimage-guide.pdf

rzr, 2023-12-17:
May this issue forwarded somewhere else, and please share your kernel versions because I've observed this on debian-12 too, and I know I had a armbian-10 that was working as expected

```
rzr@freedombox:~$ cat /proc/device-tree/model 
Olimex A20-Olinuxino Micro

rzr@freedombox:~$ cat /proc/version 
Linux version 6.1.0-13-armmp-lpae (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP Debian 6.1.55-1 (2023-09-29)
```