OLIMEX/OLINUXINO #96: Olimage-guide: suggestions

ilario, 2023-08-07:
Dear Olimex,
I am following the [Olimage-guide](https://github.com/OLIMEX/OLINUXINO/tree/master/DOCUMENTS/OLIMAGE) using a A64-OLinuXino-2Ge8G-IND with the [A64-OLinuXino-bullseye-base-20230515-130040.img.7z](https://images.olimex.com/release/a64/A64-OLinuXino-bullseye-base-20230515-130040.img.7z) image and I would suggest a few edits to the guide (and some other general issues, please point me to the right place for opening a ticket and I will open specific tickets there):

* For flashing the .img file on the SD card on Linux, I suppose one should use dd or gnome-disks, right? A word on this in the guide would be useful.
* When plugging the [USB-SERIAL-F](https://www.olimex.com/Products/Components/Cables/USB-Serial-Cable/USB-SERIAL-F/), it was clear where to plug the GND blue wire, but it was not obvious to me where to plug the Tx and Rx cable. I would add some tutorial in the guide, something like "The order is: blue wire (GND) attached to the pin close to the power plug, then red in the middle and green on the other side."
* On Windows 11, when I plugged the USB-SERIAL-F, I got a non-functioning entry in the Windows Device Manager named "THIS IS NOT PROLIFIC PL2303. PLEASE CONTACT YOUR SUPPLIER." (a ridiculous [message by Prolific](https://www.medo64.com/2022/05/really-prolific/)) and it took a while to figure out how to fix. I would suggest mentioning this message in the guide and detailing how to go on: download the 2008 driver linked at the [Olimex product page](https://www.olimex.com/Products/Components/Cables/USB-Serial-Cable/USB-SERIAL-F/), open the Windows Device Manager, right click the entry in Ports (COM&PLT), select Update driver > Browse my computer > Let me pick from a list of available drivers on my computer > select the newly installed "3.3.2.105 [10/27/2008]" one.
* The default serial connection parameters are the ones mentioned in the guide, but they do not work for A64. I had to change the Flow control from XON/XOFF to None.
* The speed (115200) is present in the screenshot included in the guide but it would be good to specify it also in the text, as it is mandatory.
* Following the guide, an optional step advised me to update the system. But when connecting the A64 to an ethernet cable with internet connection, connection is available. Actually I can see only the `lo` network interface... I just found the solution on this thread: https://www.olimex.com/forum/index.php?topic=7921.0 . The guide could mention the possible issue and point the user to the solution. I just saw that this is mentioned in the product page, but obviously I did not read it before XD
* I am not sure this is the correct place for this... In the [A64-OLinuXino-2Ge8G-IND product page](https://www.olimex.com/Products/OLinuXino/A64/A64-OLinuXino/open-source-hardware), an "8GB extra flash" is mentioned, but I cannot see in the guide how to mount this. It could be mentioned that it can be mounted with "mkdir /mnt/internal; mount /dev/mmcblk1p1 /mnt/internal". To have it automatically mounted would be ideal.
* In page 20 there is a typo: "ifconifg -a" which should be "ifconfig -a". Anyway it should be replaced with "ip address show"
* Another out-of-place comment: in the guide it is mentioned that the CPU temperature can be accessed with `cat /sys/class/thermal/thermal_zone0/temp`. It is unexpected that this is not included in the `sensors` command nor detected by `sensors-detect`.

Thanks for your amazing products!!

DanKoloff, 2023-08-22:
Thanks for the feedback. Most are good points I will look to implement, except the first one:

- using dd or gnome-disks is not recommended, users that know what they are doing don't need a guide on how to prepare an sd card for Linux, and this guide is meant for absolute beginners, so they should use BalenaEtcher or USBImager; I should expand that part of the guide tho since it is good idea to also test each card with F3 or H2testw beforehand; big chunk of all problems come from poor cards or poorly prepared cards;

ilario, 2023-08-22:
>     * using dd or gnome-disks is not recommended, users that know what they are doing don't need a guide on how to prepare an sd card for Linux, and this guide is meant for absolute beginners, so they should use BalenaEtcher or USBImager;

Yep, you are right, makes a lot of sense.

> I should expand that part of the guide tho since it is good idea to also test each card with F3 or H2testw beforehand; big chunk of all problems come from poor cards or poorly prepared cards;

Wow, good idea!
Also `badblocks -sn` or `badblocks -sw` could work for that (it could take very long time as it should do 4 passes, if a single one is enough, the `-t random` option could help).

More comments, again, please tell me if there are proper places for reporting these and I will copy them there:

* Once one soldered the UEXT pins on the A64-olinuxino board, it is not obvious which is the right direction for plugging the connector. A picture somewhere would help, for example in the UEXT guide PDF or in the A64-olinuxino quick start guide.
* The black metal box for the A64-olinuxino had zero extra space for the UEXT connector. After removing the plastic notch with a metal file (not easy without damaging the cable!) I managed to close the box with the connector plugged in. But there's quite some mechanical stress there. A metal box 1 or 2 mm wider would help.
* A recommendation of heat sink for the CPU would help. I bought this one, estimating that it would fit in the box, and it fits perfectly :D https://www.adafruit.com/product/3082
* In order to use I2C for connecting to MOD-IO2, I got very confused with i2cdetect showing random results on each launch, both with `i2cdetect 0` and `i2cdetect 1`. Finally I got that I had to activate the `sun50i-a64-i2c1.dtbo Enable TWI1 bus` overlay from `olinuxino-overlay` and then use `i2cdetect 1`. A mention on this would be useful to have on the A64-olinuxino quick start guide or on the UEXT guide PDF.

Thanks!

ilario, 2023-09-06:
This could be a security issue, and a fix should be suggested in the guide (or fixed in the OS image):

* both my A64-OLinuXino-2Ge8G-IND have the exact same SSH server key fingerprint. If this can be extrapolated to all the A64-OLinuXino-2Ge8G-IND units, it means that I could take the private key and use it for Man In The Middle attacks on SSH. I realized this when connecting to the second device:

```
$ ssh olimex@192.168.1.110
The authenticity of host '192.168.1.110 (192.168.1.110)' can't be established.
ED25519 key fingerprint is SHA256:E3xaihOuLtWT27+wajrRq9CEqDx+jCVKgdvKOLZuLtk.
This host key is known by the following other names/addresses:
    ~/.ssh/known_hosts:25: 192.168.1.111
Are you sure you want to continue connecting (yes/no/[fingerprint])?
```

And this can be confirmed logging in to both devices and running:

```
ssh-keygen -lf /etc/ssh/ssh_host_ed25519_key.pub
```

that for both devices gives:

```
256 SHA256:E3xaihOuLtWT27+wajrRq9CEqDx+jCVKgdvKOLZuLtk root@runner-gkasaxdh-project-10-concurrent-0 (ED25519)
```

ilario, 2023-09-06:
> It is good piece of info for new users. Will mention that in the manual too. But it is user's responsibility to delete or re-generate SSH keys. I'd consider that in the same league as using default user root and password olimex without changing it.

Mh, surely it has at least to be mentioned. I did not expect to have to do that. The only other similar situation I had was using OpenWrt, and its host key is generated at the first boot (the dropbear init script checks if the host key is valid and generates a valid one).

Just saw that on Raspbian there is a systemd service running only once and regenerating the host keys: https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/debian/raspberrypi-sys-mods.regenerate_ssh_host_keys.service#L8. In my opinion, using this would be a valid solution also.

ilario, 2025-01-13:
Just checked the guide, and the recommendation to regenerate the SSH keys is not there, I think this is a serious security issue and users should be strongly encouraged to act.