OLIMEX/OLINUXINO #97: A64-OLinuXino Address and Control Routing

joshua2000, 2023-09-18:
<p>why did you choose point to point routing,while the design guidelines prefer a T-branch routing?</p>

DanKoloff, 2023-09-19:
Can you point me to the exact document and exact place in it where this is recommended?

joshua2000, 2023-09-21:
[A64 PCB Layout Guide V1.0.pdf](https://github.com/OLIMEX/OLINUXINO/files/12679155/A64.PCB.Layout.Guide.V1.0.pdf)
<p>kindly check page 21,though i have checked JEDEC rules and point to point is okay.  </p> Am just asking?

DanKoloff, 2023-09-21:
It is just there is a lot of sources for DDR routing, that's why I asked no offense. We have used the document you linked, we also have it available, but as you already figured out it is a matter of preference, we just preferred point to point. Also it is usually better to pick what the memory manufacturer recommends. We mainly used these three documents as basis for the RAM routing in our devices:

1. "DDR3 Design Requirements for KeyStone Devices" by Texas Instruments (SPRABI1B)
2. "Signal Integrity and PCB layout considerations for DDR2-800 Mb/s and DDR3 Memories" by Fidus Systems Inc.
3. "Design Implementation of DDR2 / DDR3 Interfaces From a PCB Designer Perspective in Cadence Allegro" by Michael Catrambone

joshua2000, 2023-09-21:
By the way i do love the work you guys do.Am a huge fan and i thank you for your time.i completely understand your point

closed 2023-09-21 by joshua2000