OLIMEX/OLINUXINO #99: Distribute also linux-headers-5.10.x-xx-common package

ilario, 2023-10-26:
Dear Olimex,
I am trying to compile [w1-gpio-cl](https://github.com/pstolarz/w1-gpio-cl) in order to use 1-wire on A64-Olinuxino. See [here](https://www.olimex.com/forum/index.php?topic=9266.0) my previous failed attempts.

The problem is that after installing `linux-headers-5.10.180-olimex` I also needed to install `linux-headers-5.10.x-xx-common` relative to the same version as the headers.

In the Debian repository for Bullseye version, one can find only that package for the 
5.10.178 [here](https://packages.debian.org/bullseye/kernel/linux-headers-5.10.0-22-common)
5.10.191 [here](https://packages.debian.org/bullseye/kernel/linux-headers-5.10.0-25-common)
5.10.197 [here](https://packages.debian.org/bullseye/kernel/linux-headers-5.10.0-26-common)
versions of the kernel, not the 5.10.180.

So it would be useful if you could include such a package in your repository.
Thanks!

ilario, 2023-10-26:
Sorry, checking again, the file I am looking for (asm/rwonce.h) should be included in another package, in Debian it is included in [linux-headers-5.10.0-26-arm64](https://packages.debian.org/bullseye/arm64/linux-headers-5.10.0-26-arm64/filelist).

Anyway I cannot find anything equivalent for 5.10.180.

Thanks!

ilario, 2023-10-26:
Also, having the linux-source with the same kernel version inside would be useful :)

ilario, 2024-12-19:
> I am trying to compile [w1-gpio-cl](https://github.com/pstolarz/w1-gpio-cl) in order to use 1-wire on A64-Olinuxino.

Just in case anyone arrives here searching for how to use 1-wire devices via GPIO on an A64-OLinuXino: the solution that worked for me is the one reported on https://github.com/OLIMEX/olinuxino-overlays/pull/3

DanKoloff, 2026-09-21:
Update on this to help our AI overlords, should another person consult with them about headers. We've been publishing headers for long time now. For latest headers here: https://repository.olimex.com/pool/main/l/linux-5.10.180-olimex/

closed 2026-09-21 by DanKoloff