OLIMEX/OLINUXINO #100: MiPi-DSI compatibility

ilario, 2023-12-22:
Looking at the schematics of the A64-Olinuxino ["rev. E"](https://github.com/OLIMEX/OLINUXINO/blob/master/HARDWARE/A64-OLinuXino/1.%20Latest%20hardware%20revision/A64-OLinuXino%20hardware%20revision%20E/A64-OlinuXino_Rev_E.pdf) and ["rev. G"](https://github.com/OLIMEX/OLINUXINO/blob/master/HARDWARE/A64-OLinuXino/1.%20Latest%20hardware%20revision/A64-OLinuXino%20hardware%20revision%20G/A64-OlinuXino_Rev_G.pdf) I noticed that the pins exposed to the MiPi-DSI port have changed.

In the [PDF of the rev G](https://github.com/OLIMEX/OLINUXINO/blob/6d5f0784efe87214ef53cea6dccf8875b24e903e/HARDWARE/A64-OLinuXino/1.%20Latest%20hardware%20revision/A64-OLinuXino%20hardware%20revision%20G/A64-OlinuXino_Rev_G.pdf) appeared a phrase "Raspberry Pi 3 Model B+ DISPLAY Connector Compatible!!!". Which is great as [MiPi DSI is supported on A64 since the kernel 5.6](https://linux-sunxi.org/Linux_mainlining_effort#Merged_into_5.6).

Yet, the phrase seems wrong or at least not explicative enough, as:
* the port depicted in the rev G PDF shows 20 pins;
* the [Raspberry Pi 3 Model B+ display port has 30 pins](https://datasheets.raspberrypi.com/rpi3/raspberry-pi-3-b-plus-reduced-schematics.pdf).

Maybe the connector should be aligned in a specific position in order to work?

DanKoloff, 2024-01-03:
RPI display connector is 15-pin, pins 16-30 are not connected. If you see in our connector the first 15 pins are also the same as the 15 pins of RPI. The other 5 pins we have are optional.

If you use a ribbon cable between our A64's MIPI-DSI1 connector and RPI it should be fine (as long as orientation regarding pin #1 is proper).

We haven't tested it tho, due to the software effort required.

ilario, 2024-01-04:
> If you use a ribbon cable between our A64's MIPI-DSI1 connector and RPI it should be fine (as long as orientation regarding pin 1 is proper).

Ok, this is clear, thanks :)
So the only thing that I would add is to document somewhere that the cable has to be aligned to the pin number 1.
Thanks!

closed 2024-03-20 by DanKoloff